charcole/LCDZapper #1: Schematic and board creation

muramasatheninja, 2016-08-31:
Hi,

Very excited to see this project. I wasn't sure the best way to contact you so I figured I would post an issue here.

I started on converting your hand drawn schematic.
[image]

The goal is to make a self contained board that would fit in a small enclosure.

Thanks

muramasatheninja, 2016-09-02:
One thing I'm a little confused about is how the 2N7000 is used to add the reticule on the screen. I looked pretty closely at your video to see how you had it setup but wasn't able to spot it. Is there any reason to use a transistor over say just a switch to switch the video between the resistor and one that let's the signal straight through?

dlcboy, 2016-09-21:
First off just let me say this is absolutely brilliant work!!  I've been struggling for years to find a good solution to this problem and you nailed it, home run, out of the park! Bravo! I was actually buying the parts to build a similar albeit less capable system when I stumbled across your work searching to see if anybody had done this for the millionth time, and what a pleasant surprise to see it!

Seeing as how you have it done so well,  is there anything we can do to contribute and make it better? I wondered about making a logic system for the wiimote so that it can track dots actually located at the corners of the tv so there is no need for calibration and the user can move around freely? I could try putting retro reflective stickers on the tv with an ir led light source so they wouldn't need power?  I could work on sensor fusion with the wifi motion plus if you think that could be better with the help of the inertial system to increase accuracy? Your video looked pretty good already I have to say, maybe the wiimote has less noise than I expected.

Btw, muramasatheninja, imho, I suspect the 2n7000 is just being used exactly like you said, as a really fast switch... in its saturation region, just pulling the signal low briefly. Please correct if I'm wrong though!

Thanks again for the great work!! I'm anxious to get this running soon!

ysfgary, 2016-10-24:
Hi, I would like to start this project but I am using NTSC system. Any design I need to change it before I start this project? Thanks.

Catwashere, 2017-01-02:
Pretty sure 2n7000 is to remove saturation.

muramasatheninja, probably you could spot it because is attached to the input and output 
[image]

charcole, 2017-03-12:
Hi everyone,

First I've got to give a big apology. I never got a notification of this thread. I only spotted it because I'm currently working on a version 2 and wanted to look at my schematic. Thanks for the schematic muramasatheninja. I would add it but I don't want to produce any notifications at the moment as version 2 is quite close now (hopefully!).

The 2N7000 to dim is really dodgy and something I want to improve for the second version. I should say I'm mainly a programmer and not that great at analogue electronics. It worked but I don't think it was correct. The idea was to have another resistor in series with the video signal and short circuit it with the FET. When the FET was off the signal would go through the resistor causing some voltage to be dropped and therefore the display to dim. And when on it should let the signal through with only a slight drop in the FET.

However it's dodgy because the DC bias of a video signal is not set. It worked fine on the NES as it was about 3V. In more modern systems it tends to be 0V which means the signal is negative for half the time. When negative the current will flow through the diode in the FET and the gate to source voltage can go positive even when grounded. It seemed to work in this configuration but I wasn't very happy with it and the 2N7000 tended to die randomly and need replacing (too much current through the diode or static?). Also if the DC bias was high the resistor would cause the screen to dim more than if it was low.

For version 2, I was planning on improving this by putting the signal through a capacitor so I can fix the DC bias and then to an op-amp. This way the DC levels and driving circuitry is a known quantity. From there I can use two 2N7000s to switch between that output and a variable voltage source. This way the cursor can be more visible and there can be a pot to change between a white and black cursor.

In case people are interested, the other changes I was planning for V2, is replacing the Raspberry Pi and Arduino with an ESP32. It has bluetooth built in and I have it working with a Wiimote (it works a lot more reliably than with the Pi too as a bonus). The ESP32 is dual core so the other core can be used to do the precision timing needed to pick out parts of the frame (what the Arduino used to do). However the ESP32 also has a peripheral that should be able to do it without much CPU time so that's what I'm planning to use. Finally I'm going to use a 555 timer to use the dimming signal and the output from the comparator to flash the LED so the ESP32 don't need to listen to the white level like the Arduino did. Tested everything in stages, just need to go through and build it up. In all it should be a much neater product and have better tolerances than the old RPi/Arduino method.

Thanks,
Charlie

charcole, 2017-03-12:
dlcboy: Zero calibration would be good. I'm wondered about adding four IR LEDs to the corner of the TV. Problem would come if the system could only see one LED and it wouldn't know what corner it was seeing. You could try and keep track of the points (last time I saw one around here it was the top left LED). Your idea of using the accelerometer could add some more data to this to make it more accurate. Or the micro could control the LEDs and flash them in sequence if unsure which it was looking at. It's something that's on my mind but will try and get V2 working first. It'd just be software from then on.

ysfgary: Hi. NTSC would be cool. I think the only thing that'd need to change would be the timings and the line number calculations. It shouldn't need hardware changes. Might be worth waiting for V2 now though.

Catwashere: Good spotting

Catwashere, 2017-03-13:
Can't wait to see V2 and see if you convince me to buy an ESP32.

Please this time put the schematic hahahaha.

Greyneon, 2017-03-16:
Was looking at this about a week ago or so and now i hear about a version 2? 

I am SO subscribing to this!

muramasatheninja, 2017-03-24:
Thanks for the update. Definitely looking forward to Version 2!

charcole, 2017-04-03:
Another quick update for you all. Got the Version 2 working at the weekend on strip board. This is the new analogue board with the ESP32 handling Bluetooth (for Wiimote) and generating the video overlay. Video overlay stability is a lot better this time around. Next steps are to make a schematic (probably in KiCad) and layout a PCB so it can be sent off to be fabricated. Hopefully that shouldn't take too long.

[image]

Greyneon, 2017-04-10:
Awesome news! Can't wait :D

charcole, 2017-04-15:
Another update for you. So, last week I finished off laying out the PCB and sent it off to the fabricators. Expecting the first batch of boards back in the next couple weeks. From there I just need to assemble them to ensure I didn't mess anything up and improve the software side of things ready for release.
[image]

dlcboy, 2017-04-15:
Awesome work!! I'm so excited to see this moving along!!!! When it went quiet I had thought you might have moved on to something else. I tried and tried to get wii library support working on my pi and just failed, so I'm super excited to have something that's more robust and a bit more dedicated to this one task (which I will be using 100% of my free time of course;) are you making a handful of boards? Can I help in any way? I'd be happy to assemble or solder if you are doing it yourself. Also happy to help by donating enough beer to the cause to make it less painful too;)

 I can't wait to get one running to try out different sensor fusion schemes, and becon setups. I feel like it's pretty important for these old games to have dead on accuracy with minimum latency. Keep up the good work!! I can't wait to have duck hunt on my giant 2016 hdtv:D

charcole, 2017-04-16:
Thanks. Yeah, didn't have much free time the past year or so due to work but hopefully will start doing some more home projects again. I'm sure I'll have messed up something on these first batch of boards but if not I'm happy to send any left over unpopulated ones out if anyone wants one. I've just had confirmation they should get delivered on Thursday. As for completed boards, if I decide to sell some the PCB fabricators should be able to do almost all the assembly for a pretty reasonable cost but thanks for the offer. Whatever I do or not though, everything you'll need to build one will still get added to github.

Catwashere, 2017-04-17:
I'm in charcole, but I live in México. If you send me you paypal and cost I pay for the shipment.

jbp11, 2017-04-19:
This is fantastic!  I would gladly purchase a board.

charcole, 2017-04-20:
I said the boards should be here today so thought I should give another update. They actually turned up on Tuesday so I've already had time to populate one and check for mistakes. Surprisingly no major errors so they do work. I did make a mistake on the silkscreen (a resistor and capacitor label were switched). All the other problems were mechanical (screw holes too small, RCA jacks sit too high etc) but they don't matter if you aren't trying to fit it into the case I chose.

There are a couple minor issues too. For some reason the vertical position of the detection point jitters by a couple lines which I can't remember happening on my strip board version (it's not too noticeable as there's shake from holding the gun but I want to know what it is). Also one time it booted up and the APB clock seemed to be running at half rate which meant the detection point became stretched. Again, it's not a deal breaker as it came back fine after a reboot but I'd like to know what's going on there too.

As for sending boards out they are only 17 grams so should only be the cost of a letter so happy to pay postage to anywhere. Just to be clear these are just the bare PCBs with no components soldered on. You'll have to source the parts and solder everything, build the programming board (basically a couple switches and a USB to serial adapter) and set up the ESP32 tool chain. Also you might have to wait a while until I release the code and schematics as they are not quite ready yet. If you are still interested though then send me your address at cdcfoscam at yahoo.com and I'll pop them in the mail (I have 4 or 5 I probably won't use). Completed boards (just plug in and play) should be available at some point in the future if I can work out the economics of it.

[image]

rEd2k, 2017-04-23:
If you can think about possibly building the whole thing and make complete units ready to go, I'm sure people will buy them. You could sell them on websites that offer accessories for retro gaming ect. and since your the ONLY person doing this you can patent the product and be the only person out there that people can get these from making you rich! (Or maybe just more wealthy lol).

My 2 friends and I are already interested in buying a completed module, that's 3 people that will buy it, and all of the people we know that will want one as well and so on... You should really think about manufacturing these and going all the way, there isn't anyone else doing this so you have an edge. I hope you think about it and really consider what I have said, being a business owner myself I can see the potential here! Thank you for your hard work on this project so far, now it's time to take it all the way! ;)

charcole, 2017-04-23:
Thanks rEd2k. I'd certainly like to. Biggest stumbling block after reading up on the subject seems to be FCC regulations (and similar in other countries). Before selling an electronics product it needs to be tested for emissions and reading around it seems like you need to budget at least $10K (probably more like $20K) to get an intentional radiator through. The European regulations (which I'd have to adhere to) are less strict but selling something without doing the right testing could still be a huge fine or even a prison sentence. So starting small and selling a few doesn't seem like it's really an option from what I've read. I'm still investigating though. In the meantime I'm still working on the design, trying to make it as slick as possible. Currently working towards getting it to a state that a friend can beta test it.

rowie4life, 2017-04-24:
Do want.

jlebrech, 2017-04-26:
@charcole just sell as a kit, this is what edtracker did.

you might get away with people only need to solder one component or lead.

charcole, 2017-04-26:
Unfortunately kits are explicitly mentioned as still needing testing. My reading of it is selling the PCB would be fine but the PCB with any number of parts wouldn't. There's is an exemption for subassemblies which this could maybe fall into (and edtracker might) but I'd probably need to talk to someone who knows this area of law.

Other than this, I've been working away improving the circuit and trying out some ideas some friends suggested (adding IR LEDs to work as sensor bar & Wiimote-only mode). I still need to bug fix some of the problems I found (and exist on a second board I assembled). Then I'll do another revision.

[image]
(Version 2.0 board with a "hat" that'll get built into next revision)

kahhar, 2017-04-29:
Is there any possibility of adding HDMI capability?

I think it would be great to support a standard that seems to have some future lasting power--newer TVs are starting to remove composite and component inputs.

I'd love to be able to play these games on a high quality signal on my HDMI NES and the SNES, Playstation 1/2, and Dreamcast via RGB -> HDMI converters.

Catwashere, 2017-04-29:
@kahhar even if you have an RGB or HDMI mod in your consoles if you don't disable default ports with the mod you can use this, the only feature you will be missing will be the "cheat" cross/circle on the display where you are pointing

Catwashere, 2017-04-29:
@charcole Wiimote-only mode? What does that mean?

IR Led sensor bar it will by a great add on and easy to create because in last revision you need to have a costume sensor bar, a battery sensor bar or a wii connected on.

kahhar, 2017-04-29:
@Catwashere It will still work, but it will still require a composite port (and won't look as nice).

charcole, 2017-04-29:
@Catwashere Wiimote-only mode is playing light gun games without a light gun. In the picture I posted the device has a NES controller plug connected to it and it goes into the NES instead of the Zapper. Then you can play it totally with the Wiimote like a Wii game (A or B cause the trigger pull signal to be sent to the NES). It's just a little extra. I wondered if it might be an easier sell as it's slicker that way. Extra consoles could be supported with different leads (the 8-bit ones being especially easy).

@kahhar Adding HDMI would be a big change and make the device a lot more expensive and difficult to build so I probably won't, sorry. As Catwashere said if you can get composite out as well as RGB (which goes to HDMI convertor or SCART or whatever) it'll still function. The composite signal would go to the device and the RGB off to the TV (the composite output from the device would be left unconnected). You'd lose the reticule that I add but it's mainly useful for calibration and playing Wii style.

Snaid1, 2017-04-30:
I love this idea and can't wait to try it out once you get to the point you're happy with!

just a quick question, would the wiimote-only mode be an option in addition to being able to play with a light gun or would it replace the use of the gun altogether?

charcole, 2017-05-03:
As you might have noticed I've uploaded new Kicad files to a "version2" branch. It's not final (got plenty of ideas to make it better) but it matches the few boards I've sent out and the ones I've been testing with. (Nb. The R10/C3 silkscreen labels are correct in the uploaded file but were wrong on the boards I sent out so take care).

@Snaid1 Yes, the wiimote-only mode is an option. You'll still be able to play with any light gun too, it'll just be two different leads.

ddbehsnilian, 2017-06-10:
@charcole thank you for coming up with this amazing solution! I'd like to follow this project as closely as I can, is this the best place to do so?

charcole, 2017-06-10:
Hi @ddbehsnilian, Yes, this is the best place. Updates have been a bit lacking recently because I haven't had much time to work on it. Hoping to publish the code soon though. If I'm organised I will do it as three projects, this, a standalone Wiimote library and (very niche) a CD-i controller which also use ESP32 & Wiimote.

wilhouse55, 2017-06-25:
I would def be interested in purchasing an NTSC setup that requires as little work on my end as my soldering skills are non-existent!

charcole, 2017-07-01:
Had some time off this week so there's finally been some progress again! :) You should find that the firmware is now released and I've added a description of the little programming board that is needed to update the firmware. I've also updated the README with some descriptions of the new features. As version2 is now more fully featured than the old Raspberry Pi/Arduino solution I've finally merged the version2 branch into master so it'll be the first one you find when visiting the project on github.

The firmware incorporates some hardware design changes I'm planning to make but I think it should still work with the current boards. The bits that won't work are turning off the cursor with the Home button (on the old boards it was a jumper) and two player will always be in co-op mode. I can discuss how to mod the changes into existing boards if anyone wants me to.

Biggest issue left is I've still got to investigate the jitter in the image (which is more noticeable now we have some rudimentary text being displayed). Next big thing is I need to support tracking two spots so you can roll the Wiimotes. I also want to add in a logo on power up to make it a bit cooler looking. I'm also planning on removing the LM1881 from the design as they are expensive to get from proper suppliers (although dirt cheap on eBay). This will require a little bit of code changes to support (but again shouldn't break the existing boards). While I think of it, when in two player it'd be good to have different cursors too. Sure I'll think of more things to improve but the version committed should be a decent first release.

ddbehsnilian, 2017-07-02:
This is fantastic news! Are you giving any thought into adding support for NTSC signal as well as PAL? If you're planning on selling this solution that would considerably enlarge the market for it. I hope you can put whatever time you think is necessary into making this a great product.

charcole, 2017-07-02:
I have indeed. All the timing constants are grouped together in one place. As the line frequency of NTSC and PAL are very similar only TIMING_VISIBLE_LINES probably needs changing to get it to work. There are a bunch of others in the group but mainly they only need adjusting to maintain aspect ratios. The current set of constants were just eye-balled so I probably need to revisit them and set them to proper PAL timings. When I do that I'll add the NTSC timings too.

charcole, 2017-07-02:
Committed a few new changes this evening...
- Jitter problems have been much improved (Not completely fixed yet as not sure why it's happening)
- Added logo on boot
- Replaced low resolution (12 by 5 pixel) text with images
- Made code change that allows removal of LM1881 (works off composite sync only)
- Player 2 now has a different shaped reticule (a diamond)

So that leaves (on the software side of things) a proper jitter fix, tracking two spots for controller roll and finalising PAL/NTSC timings.

cr4zymanz0r, 2017-07-09:
Is there a list of what all lightguns this has been found to be compatible/incompatible with?
I'm very curious if it works with PS2 Guncons So i can play Time Crisis 2 and 3.

charcole, 2017-07-09:
It should be universal so PS2 should work fine. I have been slowly gathering up light guns and consoles to test including PS2. In my original video I think I did NES, Saturn and Dreamcast. I've now also got Master System, PS1 and PS2. I'm missing a Superscope and Menacer but I hope to be able to borrow those. I've also got a Lethal Enforcers JAMMA arcade board being delivered that I plan to test with it too. That should cover most things people will want to use with it.

cr4zymanz0r, 2017-07-09:
I greatly appreciate this project since one of the main reasons I've been unable to get rid of aging CRTs is due to needing them for lightgun games. When using RGB output from consoles paired with devices such as the Framemeister or OSSC (no lag added!) these old console look stunning on HDTVs. Lightgun compatibility with LCDs is one of the last things I need for a "perfect" setup. I don't what your final plans are for the layout of this, but I thought I'd just give my thoughts as feedback.

1.) All of my retro consoles are using RGB with csync. I'm hoping I can just feed the csync into your device and then send the output from the device to my upscaler/whatever (or not send the output and just not have extra features like the on-screen cursor?)

2.) Strapping a wiimote to each light gun is probably the simplest solution with it being easily available, but it adds more weight and isn't very elegant in my opinion. Is the wiimote being used soley for IR tracking (and bluetooth for connectivity)? I didn't know if there's some mass produced components that could be used instead and just modded into the light guns. If not, I wonder if the bare wiimote PCB is small enough to be put into lightguns (and needed buttons added externally) all while possibly being able to be powered by the voltage provided to the light gun from the console.

I realize it would be impractical for you to be modding every single light gun per order, but it could be an option for a kit such as: 
Kit A - most consumer friendly version meant for strapping a wiimote
Kit B - For more advanced users/modders that want a more elegant integrated solution.

charcole, 2017-07-09:
Yeah, the real problem is making a product enough people will want to buy to make it worth while. I remember hearing about some people who were modding Dreamcast Guns to fit Wiimote internals in them. I haven't looked into it though. Seemed like a very low volume thing. Similarly there are some old commercial IR based light-guns out there (for CD-i/PS3 etc) but they are expensive and hard to find. Most people probably have a Wiimote already (so free) although as you say not the most elegant solution.

Thanks for your comments on csync. The sync separator and video pass-through are different circuits so I could make a small modification to the board to allow you to optionally provide a separate csync signal and have the overlay added to whatever colour channel you pass through the normal composite video path. I'll experiment with that when my arcade board turns up as that'll be RGB with CSYNC. I'll then incorporate it into the next version of the board.

retrorgb, 2017-07-10:
I think there would be a ton of interest in this.  Would you be able to do a skype interview for my podcast and see what people think?  Please contact me if you're interested:  http://www.retrorgb.com/contact.html

ddbehsnilian, 2017-07-10:
Hope you accept the invite! @retrorgb does a great job maintaining the classic gaming community at its peak! His YouTube channel is really worth checking out too http://www.youtube.com/user/RetroRGB

mbrener, 2017-07-12:
Hi I'm building one of these to get two guncon 2 controllers up and running for Tim Crisis 2/3 on PS2. I'm making a few hardware changes and would like to know your thoughts. I'm going to use a NAND gate to drive the 555 timers. I'm also considering using a TXB0108 level converter to shift digital signals between 3V3 and 5V. I've been trying to decide on the video dim circuit. Does it affect the output scalling the input by 2? From what I can tell of your last posted circuit you are using two 2n7000 to disconnect the input signal and connect a ref voltage? How fast does the transition need to stabalise to work? The settling caused by switching of the fets may be contributing to your jitter. Have you considered using a high speed video mux? I found this link for a cheap mux circuit but I think the transitions are too slow (http://www.analog.com/en/analog-dialogue/articles/video-multiplexing-using-high-speed-amplifiers.html ).

charcole, 2017-07-12:
NAND gate and level shifter both sound like a sensible changes. I considered some dedicated video mux ICs but all the ones I found, in the price bracket at least, were too slow so abandoned the idea. I'm sure there are some around that'd be suitable. My switcher does cause some ringing (maybe caused by current draw on op-amp suddenly increasing). The ringing creates an unintentional but not unwelcome drop shadow on the overlay. A whole line's information is about 52 micro seconds so for one SD pixel is something like 80ns, so that's the sort of switching time we're aiming for.

The switching I don't think causes the jitter. The waveforms all look clean. Seems to instead be some interaction between the two cores of the ESP32. Once two Wiimotes are connected the jitter gets worse even though that all runs on a different core. The odd thing is it doesn't effect the horizontal sync in the same way which you'd think would be the harder of the two to get right.

BTW, For the people who are interested I've sent a message to RetroRGB about that interview.

boomchaU2, 2017-07-12:
Just came upon this as my wife is making get rid of my old CRT TV (I'll just relocate to shed, lol). This is a great idea and I will buy an NTSC version when you are ready!

dlcboy, 2017-07-12:
I've got a super scope I can lend you if you need it still.

charcole, 2017-07-13:
@dlcboy Thanks for the offer. I know someone locally with one though.

charcole, 2017-07-23:
Tiny update. Just to say I finally received the Lethal Enforcers JAMMA board I bought a couple weeks back so I've been able to test the device on that and happy to say it works. Another system crossed off the test list! :) I had no gun so wired up a two transistor interface that meant I could plug it directly into the board and play it just with a Wiimote. One new issue I found is there's no calibration settings and shots landed to the left of the cursor (the real light gun must have had a small delay between sensing and triggering). I'll see if I can think of a way of cheaply building this into the next device revision to make correcting these problems easier.

I was playing on my SuperGun with modified SCART cable. The red channel was broken out and fed through the normal composite path so I could add the overlay and the composite sync was fed directly into the LM1881N. For the next version I'll support this better (probably adding another RCA jack for composite sync). The modified SCART cable was a quite flaky (signal dropped out if I moved it) so I've designed a small breakout board and added that to this project in case it helps anyone else. Not tested yet though as waiting for it to be fabricated so be cautious if you do try it. Also RCA jacks are switching ones on that design which don't seem easy to find (or I don't know what to search for).

charcole, 2017-07-25:
I've managed to find the cause for the vertical jitter! :) Very long story short, the timer on the ESP32 doesn't seem to latch consistently unless you read the full 64 bit value (first the high 32bits then the low 32bits). The timer should only increment the upper 32-bits every 100 seconds so I didn't bother reading it in the original code. To read from the timer on the ESP32 you first have to latch it, then read the latched value. When only reading the lower bits the old value was sometimes read twice (ie. the latch didn't work or didn't happen before the read happened). This meant vsyncs were missed or extra vsyncs were detected causing the picture to jitter up and down. Adding nops between the latch and read caused it to be much more stable but pointlessly reading the upper 32-bits of the timer caused the jitter to go entirely. Couldn't find anyone else reporting it but I could believe it's a bug in the chip or the documentation.

wilhouse55, 2017-07-25:
amazing that you could find it! good job!

charcole, 2017-08-03:
I did an interview with RetroRGB about this project if anyone's interested (https://youtu.be/mVedKqHTQaA). Unfortunately the video got a bit messed up but the audio is fine.

Also the SCART breakout I've received and tested. Intercepted colour channel is coming through a bit dim which I need to look at. I've attached a photo of it working with Lethal Enforcers. 

[image]

ddbehsnilian, 2017-08-03:
Really happy to see this moving forward so steadily. It was a pleasure to watch the interview with @retrorgb Looking forward to further development of this great idea.

Catwashere, 2017-08-03:
Great interview, congrats @charcole!!!

charcole, 2017-08-23:
I've just push a version3 branch. This is the new version I've been working on. I've just sent the files off to PCBWay but it's untested so in a separate branch for now. Got a bunch of improvements, including...

 - 2 Player support
 - Built in sensor bar
 - Separate composite sync path
 - Improved the overlay and video quality (hopefully)
 - Lots more GPIO
 - GPIO should work with arcade machines / 8-bit consoles with no external components
 - GPIO ports have delay circuitry for fixing issues arcades with no calibration settings
 - Built in programmer board so only need USB to serial adapter to re-program

Was a struggle to get it all to fit!

[image]
[image]

ddbehsnilian, 2017-08-23:
I am so happy this is coming along so nicely. THANK YOU!

Catwashere, 2017-08-23:
Congratulation @charcole great to see your are making big steps

wilhouse55, 2017-08-23:
impressive. That's some nice work. I can't wait till it's ready to go.
ml

Benjamin-Carette, 2017-08-24:
This is so cool.
Hope you will release it soon. I need terminator II arcade game so badly :-)

charcole, 2017-09-10:
It's been a couple weeks so I thought I should give a new update. The new version3 boards got delivered last week and I've built one up. There were a quite a few problems with it so had to spend a few evenings debugging it. Therefore it's ended up with a bunch bodges to get all the features working. However checked it with the NES (LED and via controller port) and the Lethal Enforcers board yesterday and all the new features work now. Here's the completed board.

[image]

As you can see it's pretty busy now with ports on all sides which makes it difficult to fit it in a case. So next revision I'm planning to focus mainly on getting the form factor right. I also have some ideas of how to simplify the circuit and remove the trimmers. I'll probably update the schematic with the fixes but won't lay it out again as the footprints going to change (therefore don't use the PCB from the version3 branch if you are building your own). Next big update you see should be the new board revision.

mnbrian, 2017-09-12:
I would be interested in getting one of these if they are available. Have you thought of selling a "kit" that just has the coded chip and a list of parts?  I am in the USA, so I would be interested in the ntsc version

[333 more replies]