charcole/LCDZapper #11: Guncon2 support?

DieKatzchen, 2018-04-29:
Will it be possible to support the ps2 light gun with this? If I remember correctly the device was USB, so I'm not sure if you're able to emulate a USB device.

charcole, 2018-06-16:
I think the Guncon 2 was a fairly normal lightgun even though it had a USB connection. Should be possible to support it but would need to use the universal solution of putting an LED into the barrel of the PS2 light gun and strapping a Wiimote to the side of the gun

DieKatzchen, 2018-06-16:
I mean, it can be even simpler. The wiimote gives absolute pointer position and the Guncon 2 gave absolute mouse position.

charcole, 2018-06-16:
Yeah, it'd definitely be possible to do it that way. You'd need some extra circuitry to mimic the USB device though. I'm planning on adding a serial output option for people that want to build external circuitry to interface to other systems directly. CDi is like that too (not that there's many good games on that system).

DieKatzchen, 2018-06-16:
That's probably easily done. Could probably even do it with a digispark or trinket.

charcole, 2018-07-03:
Serial output should be available by setting the IOType setting to Serial. This should enable this use. It currently outputs 0x80 (upper bit set to detect start byte), player number, upper 7 bits of X, lower 7 bits of X, upper 7 bits of Y, lower 7 bits of Y, upper 7 bits of buttons, lower 7 bits of buttons.

closed 2018-07-03 by charcole