charcole/LCDZapper #16: Build Log

arithmaldor, 2018-09-02:
I'm going to try my hand at building one of these. I am logging my progress [here](https://tinkerplunk.com/lightgunverter-build-log/). If anyone sees any issues or things I get wrong please let me know, and @charcole has been gracious enough to answer my questions. I will post any further questions here as I go.

arithmaldor, 2018-09-02:
So are the 4 LEDs on the front supposed to be IR or are they for decoration? I don't know how they'd function as a sensor bar without being spaced out, or can the Wii remote function with only one point of reference?

arithmaldor, 2018-09-02:
Found the answer
https://github.com/charcole/LCDZapper/issues/1#issuecomment-341992825
Only needs location, not rotation, so it works with one spot

charcole, 2018-09-02:
Yes, sorry you beat me to it. I've been thinking recently about adding better software support for standard light bars too but I've not had much problem with the single reference point.

The LEDs needs to be IR as you worked out. They just work together to form one big spot. I'd suggest using 940nm IR LEDs as I think the Wiimote is more sensitive to them. Also get ones with the biggest spread you can find so it'll work from more angles.

arithmaldor, 2018-09-16:
PCB fab takes a while, should have gone with express shipping from jlcpcb. It is so funny to me that the pcb files specifically say "manufacturere marks on this side please" and they still put them on the outside. Front and back panels are the only ones here, main board should be here next week and I can start building. @charcole is there any documentation on the cables? I have all of the systems supported or soon to be supported by just wii remote except saturn (sold sadly). I think I have an ethernet crimper around here somewhere...

charcole, 2018-09-17:
Yeah, I normally have to specifically request manufacturer's marks on the rear silkscreen. Only used pcbway but they've done followed the instructions every time so far. As you saw I also added it on the silkscreen to try and make extra sure.

There isn't good documentation on cables at the moment but I'm hoping to get that done soon. I've got to make some guides for the person manufacturing the commercial cables so I should be able to put the same documentation up.

arithmaldor, 2018-09-27:
Finished the build, just waiting on a programmer. I've got some other projects going on so I'm not in a big rush to build the cables, I'll make them whenever the guides are ready. Turned out really well I think.

charcole, 2018-09-27:
Awesome! Well done, Jacob! I’ll get the cable plans ready as soon as I can. Hope you enjoy it when it’s finished!

arithmaldor, 2018-10-15:
Hey @charcole I've got the board assembled but I'm not experienced with programming or compiling the firmware. I downloaded the toolchain, pointed it at the folder, and used the make command but got nothing. I'll have to look and see exactly what error I got, but do you have any tips?

Also, [this](https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F253056283240) is the programmer I got, but I'm not exactly sure which lines to hook up or if this is the right kind.

SuperVehicle-001, 2018-11-09:
Hi @charcole, thank you for making your project open source (non-commercial).  I think my board is mostly functional.  It displays the logo overlay w/ composite in/out setup.  I messed around a little with csync input for RGB setup, but really have no idea how to proceed from here with pairing wiimotes and building cables.   Thanks @arithmaldor for the blog write-up, would not have attempted without it.

[image]

arithmaldor, 2018-11-09:
Awesome work @SuperVehicle-001 ! Any tips on programming? That's where I stalled as I have next to no experience with that.

Heath3827, 2019-07-27:
Do you have more concise instructions on how to build? A proper diagram and parts list? I'd like to build one but as a hobbyist (not an engineer or programmer) I would need more monkey see - monkey do instructions, preferably not using surface mount components (I have no experience soldering such tiny components). Otherwise is there the ability to acquire a unit? I just want to play Duck Hunt and Metal Combat again.

arithmaldor, 2019-07-31:
my build log is about as step by step as you're going to get, and the board is designed with surface mount components, so you're not going to get through hole parts without redesigning the whole thing. Can be reserved at lightgunverter.com

IDBurner, 2019-08-13:
@arithmaldor Did you ever have luck with the programmer?

Also can anyone point me in a direction to make a Dreamcast Cable for this?

pattonthompson, 2019-09-05:
Arithmaldor your build guide is awesome.  I can handle the soldering.  I too am not sure about the programming.  Can anyone provide instructions on how to do the programming and flash the firmware?  Would be greatly appreciated!

dekkit, 2020-11-30:
Are there any recommended tools to compile the fireware code / get the toolchain sorted.   When i get a bit of time i wouldnt mind having a crack at adding the psx serial settings.

Any links to tutorials /software would be most helpful.

Also is there a factory default .bin in case i bork it that i can manually flash back on?

charcole, 2020-12-01:
The toolchain should be the standard ESP32 IDF (https://docs.espressif.com/projects/esp-idf/en/stable/get-started/). Has been a while so might be some issues with the newest version but hopefully nothing too difficult to solve. I can find the exact revision I used if you have problems. If it asks a bunch of questions making for the first time you can probably just accept the defaults as they will all be new features added to the SDK since I last used it.

The programming over WiFi is easiest but it is also easy to brick if you only have that method as the new firmware has to work well enough to trigger and run the firmware update code. The create_image.sh script is what turns the linkers output into a file that the LightGunVerter firmware updater accepts (it just appends LGV_FIRM to the start).

To mitigate the risk of bricking I'd suggest getting a ESP32 dev board like this (https://ebay.us/yJ2HjH) so you can program using the serial 'make flash monitor' method. Ensure this new firmware can go through the WiFi update process without bricking before writing it to your LightGunVerter.

To reprogram a bricked device (or to program a fully built one for the first time) you need to do so via serial. This is where it gets a bit complicated. I used a serial to USB adapter like this (https://www.ebay.co.uk/itm/USB-To-TTL-FT232RL-FTDI-Serial-Adapter-Converter-Module-For-Arduino-3-3V-5V-Mini/222727769676) and you can just stick it in the programming holes on the board at an angle (or solder a header on if you want it more secure). You do have to get it the right way round of course (I'd check the GNDs match). Finally there's the problem that the 3.3V supply of that module can't supply the whole board so I also added a beefier 3.3V regulator to it. The inputs for the regulator coming from 5V and GND on the programmer and the output going to the middle pin of the jumper (with the jumper removed). You might also be able to do it by removing the jumper and then powering both the board and the serial adapter at the same time from the same USB port (via a passive hub). Been too long for me to remember if that is a good idea to do or not. Finally you need to trigger the serial update process. For this there are two buttons (reset and upload) on the board. On the production board these might have been unpopulated to save cost. The upload button needs to be held while you press reset. Then go to your computer and run 'make flash monitor' and that should connect, reprogram the flash, reboot and then show you the boot TTY. You might need to do 'make erase_flash' if you've already updated via WiFi before programming otherwise the other partitions might still have the old firmware.

In the interests of sharing knowledge, for production I used a programmer like this (https://www.ebay.co.uk/itm/Programmer-Tool-ESP32-Adapter-Socket-Kit-For-ESPRESSIF-ESP-WROOM-32-Module-OS/373373097299) to program all the ESP32 modules and test them before soldering to the board and from then on used WiFi for any last minute patches. Really love these programmers. They act just like the dev boards but you can easily add and remove bare ESP32 modules. Did save me making some duff boards because even getting ESP32 modules from a reputable supplier like Mouser I ended up with a couple duds.

dekkit, 2020-12-07:
Awesome thanks, much useful info.  Ill see how i go!

dtran86, 2021-01-05:
Would someone here be able to program the lightgunverter to work with this system instead of a wiimote. It's similar to how the wiimote tracks coordinates with the same ir cam, but 1:1 and no calibration needed. http://forum.arcadecontrols.com/index.php?topic=161189.0