charcole/LCDZapper #19: Nes Zapper Cable

dekkit, 2020-11-23:
Documenting examples of each cable wiring  (in case anyone needs to create additional nes ones, build their understanding of how the system works). Ive included a few photos and original schematic (as drawn by charcole)

[image]
[image]

[image]

charcole, 2020-11-23:
Thanks Dekkit, this is great. To share some more details. They were made from NES controller extender cables. These are only wired for a joypad so I had to crack open the shells and move the Strobe/Data pins over to the other side of the socket (to become Photosensor/Trigger) then glue the shell back shut. Then the female connector was snipped off and the wires crimped in the right order into the RJ45 jack.

Looks like the diagram I showed you is wrong. The photosensor and 5V lines need to be switched (at the NES jack).

dekkit, 2020-11-24:
Thanks for confirming!  

I did a nes zapper conversion a few years back from a famicom dsub15pin to nes socket and had to do example the same! (pull the pins out and reconfigure a cheap nes extension cable).

Ill draw this up properly but combine your comments and info from here i used ages ago.
http://nesdev.com/zapper_to_famicom.txt