charcole/LCDZapper #21: compile and upload the program

Verni82, 2021-02-17:
Hello, I've tried a lot of ways and unfortunately haven't managed to compile the software and flash it onto the chip. I hope someone can help me.

1. try: quick way
 I flash the ESP32 with the help of the Arduino(1.8.13) IDE example: OTA Web update in order to use the existing Firmware file. When loading the file, the error message: Wrong Magic Byte appears.

2. try long way
I installed Ubuntu on a VM and followed the steps according to the instructions. (I installed Ubuntu on a VM and followed the steps according to the instructions.) After the successful test with the Hello World example, I wanted to create and upload the firmware. I continued with the instructions from the Programmer.md file. I loaded the existing SKConfig and started make flash.
Ich denke hier gibt es ein Paar probleme mit den Bibliotheken. 
`XX build/main/main.o
In file included from /home/alex/esp/Firmware/main/main.cpp:33:
/home/alex/esp/esp-idf/components/esp_event/include/esp_event_loop.h:2:2: warning: #warning "esp_event_loop.h is deprecated, please include esp_event.h instead" [-Wcpp]
 #warning "esp_event_loop.h is deprecated, please include esp_event.h instead"
  ^~~~~~~
In file included from /home/alex/esp/Firmware/main/main.cpp:41:
/home/alex/esp/esp-idf/components/esp32/include/rom/rtc.h:1:2: warning: #warning rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead [-Wcpp]
 #warning rom/rtc.h is deprecated, please use esp32/rom/rtc.h instead
  ^~~~~~~
In file included from /home/alex/esp/Firmware/main/main.cpp:42:
/home/alex/esp/esp-idf/components/esp32/include/rom/cache.h:1:2: warning: #warning rom/cache.h is deprecated, please use esp32/rom/cache.h instead [-Wcpp]
 #warning rom/cache.h is deprecated, please use esp32/rom/cache.h instead
  ^~~~~~~
/home/alex/esp/Firmware/main/main.cpp: In function 'void WiimoteTask(void*)':
/home/alex/esp/Firmware/main/main.cpp:699:8: error: 'UART1' was not declared in this scope
    if (UART1.status.txfifo_cnt == 0 && UART2.status.txfifo_cnt == 0)
        ^~~~~
/home/alex/esp/Firmware/main/main.cpp:699:40: error: 'UART2' was not declared in this scope
    if (UART1.status.txfifo_cnt == 0 && UART2.status.txfifo_cnt == 0)
                                        ^~~~~
/home/alex/esp/Firmware/main/main.cpp:736:8: error: 'UART1' was not declared in this scope
    if (UART1.status.txfifo_cnt == 0) // UART FIFO is zero
        ^~~~~
/home/alex/esp/Firmware/main/main.cpp: In function 'void WifiInitAccessPoint()':
/home/alex/esp/Firmware/main/main.cpp:1684:21: warning: 'void tcpip_adapter_init()' is deprecated [-Wdeprecated-declarations]
  tcpip_adapter_init();
                     ^
In file included from /home/alex/esp/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from /home/alex/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/alex/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/alex/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/alex/esp/Firmware/main/main.cpp:32:
/home/alex/esp/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:34:6: note: declared here
 void tcpip_adapter_init(void)  __attribute__ ((deprecated));
      ^~~~~~~~~~~~~~~~~~
/home/alex/esp/Firmware/main/main.cpp:1684:21: warning: 'void tcpip_adapter_init()' is deprecated [-Wdeprecated-declarations]
  tcpip_adapter_init();
                     ^
In file included from /home/alex/esp/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from /home/alex/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from /home/alex/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/alex/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/alex/esp/Firmware/main/main.cpp:32:
/home/alex/esp/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:34:6: note: declared here
 void tcpip_adapter_init(void)  __attribute__ ((deprecated));
      ^~~~~~~~~~~~~~~~~~
In file included from /home/alex/esp/esp-idf/components/esp_timer/include/esp_timer.h:45,
                 from /home/alex/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:46,
                 from /home/alex/esp/esp-idf/components/freertos/include/freertos/portable.h:52,
                 from /home/alex/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:64,
                 from /home/alex/esp/Firmware/main/main.cpp:23:
/home/alex/esp/Firmware/main/main.cpp:1685:57: warning: 'esp_err_t esp_event_loop_init(system_event_cb_t, void*)' is deprecated [-Wdeprecated-declarations]
  ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
                                                         ^
/home/alex/esp/esp-idf/components/esp_common/include/esp_err.h:119:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
In file included from /home/alex/esp/esp-idf/components/esp_event/include/esp_event.h:26,
                 from /home/alex/esp/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from /home/alex/esp/Firmware/main/main.cpp:32:
/home/alex/esp/esp-idf/components/esp_event/include/esp_event_legacy.h:239:11: note: declared here
 esp_err_t esp_event_loop_init(system_event_cb_t cb, void *ctx) __attribute__ ((deprecated));
[... 141 lines cut ...]
/home/alex/esp/esp-idf/components/soc/esp32/include/soc/dport_access.h:154:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 RT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                            ^~~

/home/alex/esp/esp-idf/components/soc/esp32/include/soc/dport_access.h:183:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
/home/alex/esp/Firmware/main/main.cpp:1932:3: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
   DPORT_SET_PERI_REG_MASK(DPORT_APPCPU_CTRL_A_REG, DPORT_APPCPU_RESETTING);
   ^~~~~~~~~~~~~~~~~~~~~~~
/home/alex/esp/Firmware/main/main.cpp:1932:52: note: suggested alternative: 'EXT_CPU_RESET'
   DPORT_SET_PERI_REG_MASK(DPORT_APPCPU_CTRL_A_REG, DPORT_APPCPU_RESETTING);
                                                    ^~~~~~~~~~~~~~~~~~~~~~
/home/alex/esp/esp-idf/components/soc/esp32/include/soc/dport_access.h:154:88: note: in definition of macro '_DPORT_WRITE_PERI_REG'
 RT_WRITE_PERI_REG(addr, val) (*((volatile uint32_t *)(addr))) = (uint32_t)(val)
                                                                            ^~~

/home/alex/esp/esp-idf/components/soc/esp32/include/soc/dport_access.h:183:46: note: in expansion of macro 'DPORT_WRITE_PERI_REG'
 #define DPORT_SET_PERI_REG_MASK(reg, mask)   DPORT_WRITE_PERI_REG((reg), (DPORT_READ_PERI_REG(reg)|(mask)))
                                              ^~~~~~~~~~~~~~~~~~~~
/home/alex/esp/Firmware/main/main.cpp:1932:3: note: in expansion of macro 'DPORT_SET_PERI_REG_MASK'
   DPORT_SET_PERI_REG_MASK(DPORT_APPCPU_CTRL_A_REG, DPORT_APPCPU_RESETTING);
   ^~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [/home/alex/esp/esp-idf/make/component_wrapper.mk:292: main.o] Fehler 1
make: *** [/home/alex/esp/esp-idf/make/project.mk:635: component-main-build] Fehler 2`

Verni82, 2021-02-17:
3.try 
I downloaded the current Tasmota software for the ESP32 and started flashing with FLASH_DOWNLOAD_TOOLS_V3.4.9.2. After all partitions were entered in the tool (boot loader, ...), I replaced the Tasmota.bin with the Firmware.bin. Unfortunately, nothing came of this experiment either

charcole, 2021-02-17:
Hi Verni82,

I talked about this a little bit in the Building Discussion issue (https://github.com/charcole/LCDZapper/issues/16#issuecomment-736833128).

The reason you are getting the deprecated warnings/errors is because the ESP-IDF is always getting updated and as the code hasn't been worked on for a while it's now out of date. You could try and fix them yourself or grab an older version of the IDF. The revision I have is 3b92e85b308578c24f14f57371498303c198c7cb (and the tag was v3.1.1). The code should compile fine with the correct version.

The firmware.bin included in the release is meant for self-updating the commercial releases over WiFi. It's the plain binary just with "LGV_FIRM" appended to the start as a (rather crappy) test that its actually firmware for the LightGunVerter before flashing it. If you remove those 8 bytes at the start you may be able to program using some other method (although I have never tried or know anything about either of the ones you mentioned). I'd think the easiest route would be using the correct IDF and flashing it in the usual way (via 'make flash').

Charlie

Verni82, 2021-02-18:
Hi Charlie,
Thanks for Help :-)

I tested version 3.1.1. But there are problems on the ESp, the firmware there constantly restarts with an error message.
Compiling worked with version 3.3.1, but I'm not sure if everything works. I am currently not getting a picture via video.

charcole, 2021-02-18:
What’s the message? Did you try ‘make erase_flash’? If you do ‘make flash monitor’ after what do you get?

Verni82, 2021-02-18:
Hi Charlie,
I followed the start process with serial monitor via Arduino IDE:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5080
load:0x40078000,len:10596
load:0x40080400,len:5716
entry 0x400806c0
[0;32mI (302) cpu_start: Pro cpu up.[0m
[0;32mI (302) cpu_start: Application information:[0m
[0;32mI (302) cpu_start: Project name:     app-template[0m
[0;32mI (305) cpu_start: App version:      1[0m
[0;32mI (309) cpu_start: Compile time:     Feb 18 2021 21:22:28[0m
[0;32mI (316) cpu_start: ELF file SHA256:  c6771bf70290348e...[0m
[0;32mI (321) cpu_start: ESP-IDF:          v3.3.1-dirty[0m
[0;32mI (327) cpu_start: Starting app cpu, entry point is 0x400811f0[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (337) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (344) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (350) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (356) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (362) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (369) heap_init: At 3FFCD6A8 len 00012958 (74 KiB): DRAM[0m
[0;32mI (375) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (381) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (387) heap_init: At 40096E38 len 000091C8 (36 KiB): IRAM[0m
[0;32mI (394) cpu_start: Pro cpu start user code[0m
[0;32mI (188) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (189) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (197) gpio: GPIO[14]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (206) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (216) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (225) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (234) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (243) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (253) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (262) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (271) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (281) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
WiimoteTask running on core 0
[0;32mI (305) BTDM_INIT: BT controller compile version [5aed448][0m
[0;32mI (305) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
SpotGeneratorTask starting on core 1
[0;32mI (316) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m

I'll try to delete it with the flash tomorrow.

Verni82

charcole, 2021-02-18:
I can’t remember the start up being so verbose but that all looks fine. Seems to have started up OK. I suspect it should be working.

No. It can’t produce a video signal itself. It only overlays on top of the input video. If that helps at all.

Verni82, 2021-02-23:
Hi Charlie,
I created a new VM with Ubuntu 14 and got ESP-IDF3.1.1 to run on it. 
I had to erase the flash from the ESP32. Now the ESP also starts with the 3.1.1 version. My Wii controller is also recognized when I press 1 + 2. There are only problems with the image transmission. The signal is not passed on from Video IN to Video Out. Maybe something is broken. Thank you so much for the help. I've already come a big step further.

Verni82

charcole, 2021-02-23:
No problem. Let me know once you've got it all running. Always great to see someone else build one.

If you are getting no video then first check DIM_INV is high (or at least mainly high). If it's low that basically shunts the source video to ground so that DIM and DIM1 can replace it with the overlay. If DIM and DIM1 were playing up I'd still expect you to see something (the overlay would just be additive but it could mess with the sync I guess). If you have an oscilloscope then checking TP3, either side of R19 and R20 and TP7 should get you to where the signal is going missing.