charcole/LCDZapper #22: PSX Cable (using Arduino Nano)

dekkit, 2021-05-12:
...Work in progress...more soon

Initial place to upload some wiring diagrams, code edits and instructions on how to build an adapter cable to connect your Wii mote to your PS1/PSX using LCDGunverter.

What you'll need:
- Psx cable with 1x male socket (ie ps1/2 control pad cable extension or old control pad cable from broken controller)
- Cat 5 network cable (with an RJ45 socket)
- Arduino Nano (+ a mini usb cable to connect to pc to program)

Optional - Arduino prototyping /breadboard wires.

Note: You should be able to build this without soldering and using just cable/wire joiners (if your nano comes with pins already soldered on)

**Setting up the Ardunio Nano**
- download Arduino IDE (add links) and drivers
- ... Tbc

**Compile and upload Guncon.ino to Nano**
.....include code update to switch guncon mode and fix compiling issue.

**Wiring PSX to Ardunio**
..
.

**Wiring Arduino to LCDGunverter**
...

**Setting up LCDGunverter to use adapter**
...

**Testing the results**
Fingers crossed

dekkit, 2021-05-16:
@charcole 
Can you recall what is happening at the following lines of code with the x and y gun position?

[image]

Im wondering if there is a pal/ntsc issue with how many lines the nano guncon45 is responding back to the psx.

Also can you recall how many wii remote buttons are being communicated through to the io serial (assuming just A and B?)

dekkit, 2021-05-21:
Further testing...

The Wii mote's Dpad LEFT and RIGHT function as the guncon's A and B button.

Useful to get through game menus.

Was able to get the arduino nano to provide debug info (via adding serial print lines to the .ino filr) and viewing via the Arduino IDE serial terminal window on PC.  Outputs to ps1 seem to be correct from probing the reply outputs.

I still cant explain why the X and Y doesnt seem to work well at all (seems to be certain large chunks of screen area which dont register). This results in the screen flashing as if a normal guncon is pointed away from the screen.

Ill need to probe further with adding further debug lines to my .ino.  I hope the arduino isnt introducing a delay that is throwing out possible video timings but too early to say.

charcole, 2021-05-22:
It shouldn't be a timing issue. GunCon guns do the timing internally which is why they need the composite input (only Konami gun games needs the timing circuitry of LGV). The Arduino sketch just needs to tell the console the coordinates of the cursor. The "Serial" mode on LightGunVerter just sends the coordinates it gets from the Wiimote and the GunX/GunY math converts it to what the PSX expects. From memory I think the Wiimote returns coordinates are 0 to 1024 (X) and 768 (Y) while PSX will expect a time delay for X and a line number for Y. The LGV could return the exact line number, that's what I did for the GunCon 2 cable, but in this version I just did some maths.

X axis is flipped (1023-X). Then scaled ((FlippedX*3)>>8). >>8 (half 8 times) is same as divide by 256 so scaling is 3/256. Then 0x4D (77 in decimal) offsets it left and right. Similarly for Y 11/52 scaling plus 32. This would have been done on PAL so the Y will definitely be wrong as PAL has more lines. X should still be close but might need some tweaking.

Hope that helps.

PS. The button bit flags are shown in https://github.com/charcole/LCDZapper/blob/master/Firmware/main/esp_wiimote.h. You should get everything but the Home button.

dekkit, 2021-05-23:
Awesome thanks for the additional insights, its been a great learning experience working arduino, spi, and seeing how button switches polled and then get encoded.  Hopefully documenting as i go will help others too.

dekkit, 2021-06-05:
Finally success for GunCon45!

Added the . ino that works for me to the first post.

Actual wiring used (updated diagram)
[image]

Shortened cable, completely removed old control pad wire.

The bad cable was causing all sorts of weird effects.  The benefit however was really understanding how your code works and learning how the ps1 communicates with its control pads via SPI.

Thanks @charcole - it works great now!

dekkit, 2021-06-05:
Finally a working cable
[image]

Exposed socket, cut wire, and soldered in wires to connect to nano!
[image]

Point Blank 3 still so much fun!
[image]