charcole/LCDZapper #27: any good news about more game console's cable

hpx1984, 2024-06-12:
Hi Sir Charcole
i saw your greate project name of "LCDZapper" and got to know it can support so many old game consoles
but i just saw some informations of the Playstation cable from github ,
could you share more other console cable's information?
i want to try to build it
sorry for trouble you and many thanks your greate project LCDZapper to help all game console lover to can go ahead to play lightgun game on new LCD TV

NES / Famicom
Master System
Saturn
Playstation
Playstation 2
Dreamcast
Lethal Enforcers (Arcade)
original xbox
xbox 360
ps3/ps4
3DO
PCE
SNES
atari2600

dekkit, 2024-09-14:
If you scan through the issues, I've attempted to document how a few of the cables were designed / made - so you can build your own if you're keen.

hpx1984, 2024-09-15:
> Sent Time:2024/9/14 13:42
>
>
> Cc recipient:"hpx1984"< ***@***.*** &gt;;"Author"< ***@***.*** &gt;;

Dear Sir

yes ,im already review all the issues and got some detail of the PS cable and nes cable 
im going to try to build one 

so many thx your great project on the lighgun 

xox
***@***.***

                       
Original Email